RollingGecko/VescUartControl #7: getting additional values with VescUart

dodolbysuround, 2017-01-16:
Hi,
I'm wondering if it is possible to get VESC temperature and raw ppm information via VescUart Library.
Another question: is it possible to get the datas from a second VESC connected to the first one through CAN?
I mean:
 VESC1 (CAN id 0) ===can=== VESC2 (CAN id 1) ===uart=== ARDUINO
Thank you by advance for the answers.

RollingGecko, 2017-02-13:
Sorry for the late reply. Didn't see it.
It is possible to get the tempreature. You find it in the struct:
typedef struct {
	float v_in;
	float temp_mos1;
	float temp_mos2;
	float temp_mos3;
	float temp_mos4;
    float temp_mos5;
    float temp_mos6;
    float temp_pcb;
    float current_motor;
    float current_in;
    float rpm;
    float duty_now;
    float amp_hours;
    float amp_hours_charged;
    float watt_hours;
    float watt_hours_charged;
    int tachometer;
    int tachometer_abs;
    mc_fault_code fault_code;
} mc_values;

An adaption on the bool ProcessReadPacket(uint8_t* message, bldcMeasure& values, int len) procedure is required. I skipped there only reading the relevant bits. You must also adapt in the local_datatypes.h the struct bldcMeasure. Than you have the data. You are welcome to adapt the code and send me a pull request.

For two VESCs I have at the moment no idea. Sorry. 

Regards

Andy

dodolbysuround, 2017-02-13:
Thank you for your answer. I actually understood part of the code by myself and managed to get temperature.
I would love to share it but I'm really not familiar with GitHub and have no idea about how to send you a pull request... Wouldn't you mind and spend few minutes to explain how to do that?
Ok for two VESC.
Would it be possible to get the rough ppm signal?
Seeing the mc-values structure, I'm wondering if the fault codes are also available doing the same manipulations than for the temperatures?
Thanks again for sharing your work.
Regards
Loic

RollingGecko, 2017-02-13:
In the current version it is not that easy to get other packages read than the mc_values. It needs a redesign. It was only used for my remote project. I started in the development branch with a redesign. But I stopped it because I already started with other projects and have not the time to proceed this project. Sorry for.

closed 2017-02-13 by RollingGecko