RollingGecko/VescUartControl #11: Update for new VESC4 Firmware incl w/ Beta VESC-Tool

jeffnyc, 2017-06-20:
Now that the VESC6 is in Beta, the new VESC-Tool is out (Beta) and it includes new and improved firmware for the VESC4's.  According to Vedder this new firmware requires some changes to UART.  Can this library be modified to accommodate these changes?  Or perhaps a new VescUartControl version is needed?  Here is the summary (from Vedder) of what's required for the new firmware to work with VescUartControl:

"  
The SetRpm and SetCurrent commands should work as before, but the GetValue command probably has to be updated. This is the way the values are sent in the latest firmware:

> case COMM_GET_VALUES:
>         ind = 0;
>         send_buffer[ind++] = COMM_GET_VALUES;
>         buffer_append_float16(send_buffer, mc_interface_temp_fet_filtered(), 1e1, &ind);
>         buffer_append_float16(send_buffer, mc_interface_temp_motor_filtered(), 1e1, &ind);
>         buffer_append_float32(send_buffer, mc_interface_read_reset_avg_motor_current(), 1e2, &ind);
>         buffer_append_float32(send_buffer, mc_interface_read_reset_avg_input_current(), 1e2, &ind);
>         buffer_append_float32(send_buffer, mc_interface_read_reset_avg_id(), 1e2, &ind);
>         buffer_append_float32(send_buffer, mc_interface_read_reset_avg_iq(), 1e2, &ind);
>         buffer_append_float16(send_buffer, mc_interface_get_duty_cycle_now(), 1e3, &ind);
>         buffer_append_float32(send_buffer, mc_interface_get_rpm(), 1e0, &ind);
>         buffer_append_float16(send_buffer, GET_INPUT_VOLTAGE(), 1e1, &ind);
>         buffer_append_float32(send_buffer, mc_interface_get_amp_hours(false), 1e4, &ind);
>         buffer_append_float32(send_buffer, mc_interface_get_amp_hours_charged(false), 1e4, &ind);
>         buffer_append_float32(send_buffer, mc_interface_get_watt_hours(false), 1e4, &ind);
>         buffer_append_float32(send_buffer, mc_interface_get_watt_hours_charged(false), 1e4, &ind);
>         buffer_append_int32(send_buffer, mc_interface_get_tachometer_value(false), &ind);
>         buffer_append_int32(send_buffer, mc_interface_get_tachometer_abs_value(false), &ind);
>         send_buffer[ind++] = mc_interface_get_fault();
>         commands_send_packet(send_buffer, ind);
>         break;

"

RollingGecko, 2017-06-21:
@jeffnyc:
Just checked Vedders repo. Current FW is 2.18. There are no real changes in the command.c visible for me. The last commit is nearly a year ago on this repo. Did I miss something?

jeffnyc, 2017-06-21:
Sorry for the confusion - this is from the new firmware that is part of the Beta test program for the new Vesc 6.  As part of the Vesc 6 launch, Benjamin Vedder created the new VESCTool application.  And in addition to being the default interface for the new Vesc 6 boards, it also includes new firmware for the existing Vesc 4's that are already on the market (see screenshot attached).  And that is where the changes in UART protocol come into play...

 It hasn't been released publicly yet as it's still in Beta, which I suspect is why it's not on Vedder's repo?.  I think it is due to be released publicly sometime in July, once the Beta program has concluded.  Benjamin sent it to me to test as we are a vendor using Vesc4's in our production product that is about to launch (www.shareroller.com), and the new firmware has important improvements in motor detection and FOC operation.

When Benjamin sent me the VESCTool application, he included the note about the new UART code that I posted into the Issue on Github, as well as the comment:
"I think you can update the library you are using easily to work with this."   

 But unfortunately I cannot!  I really don't understand the code well enough to do so...
Does that make sense now?  It's possible that I'm misunderstanding things too - I hope not!
Thanks,Jeff

RollingGecko, 2017-06-21:
Please send me the complete commands.c and .h to check it completely.

jeffnyc, 2017-06-21:
Unfortunately, that's all Benjamin sent me in the email, which leads me to believe that's the only relevant change for your library?  I did tell him I was using VescUartControl and needed to know what changes might be required for my code (and if the delay could be shortened), and this is exactly what he wrote:

"The VescUartControl library is not written by me, so I'm not sure what the delay does. I usually don't use any delay and just wait for the result after sending the request. You could try increasing the baud rate and see if it helps.
  The SetRpm and SetCurrent commands should work as before, but the GetValue command probably has to be updated. This is the way the values are sent in the latest firmware:
 
 
case COMM_GET_VALUES:
         ind = 0;
         send_buffer[ind++] = COMM_GET_VALUES;
         buffer_append_float16(send_buffer, mc_interface_temp_fet_filtered(), 1e1, &ind);
         buffer_append_float16(send_buffer, mc_interface_temp_motor_filtered(), 1e1, &ind);
         buffer_append_float32(send_buffer, mc_interface_read_reset_avg_motor_current(), 1e2, &ind);
         buffer_append_float32(send_buffer, mc_interface_read_reset_avg_input_current(), 1e2, &ind);
         buffer_append_float32(send_buffer, mc_interface_read_reset_avg_id(), 1e2, &ind);
         buffer_append_float32(send_buffer, mc_interface_read_reset_avg_iq(), 1e2, &ind);
         buffer_append_float16(send_buffer, mc_interface_get_duty_cycle_now(), 1e3, &ind);
         buffer_append_float32(send_buffer, mc_interface_get_rpm(), 1e0, &ind);
         buffer_append_float16(send_buffer, GET_INPUT_VOLTAGE(), 1e1, &ind);
         buffer_append_float32(send_buffer, mc_interface_get_amp_hours(false), 1e4, &ind);
         buffer_append_float32(send_buffer, mc_interface_get_amp_hours_charged(false), 1e4, &ind);
         buffer_append_float32(send_buffer, mc_interface_get_watt_hours(false), 1e4, &ind);
         buffer_append_float32(send_buffer, mc_interface_get_watt_hours_charged(false), 1e4, &ind);
         buffer_append_int32(send_buffer, mc_interface_get_tachometer_value(false), &ind);
         buffer_append_int32(send_buffer, mc_interface_get_tachometer_abs_value(false), &ind);
         send_buffer[ind++] = mc_interface_get_fault();
         commands_send_packet(send_buffer, ind);
         break;
 I think you can update the library you are using easily to work with this.
 "
If you need me to ask him for the files you mention below or for other clarification, I'm happy to do so - was just trying not to bug him too much given how busy he is with the VESC6 Beta AND the fact that he broke his ankle badly recently and is recovering.
Thanks!-Jeff

jeffnyc, 2017-06-22:
Fyi I didn't realize that my replies were being echoed in the Github comments.  Sorry, I'm not too familiar with how Github works..
Went ahead and deleted a few things that weren't relevant for the Github repository.

RollingGecko, 2017-06-27:
OK. Let's go. Please test this branch:

https://github.com/RollingGecko/VescUartControl/tree/VESC6

RollingGecko, 2017-07-02:
Tested with a Mega2560 on VESC 4.11 and the newest firmware.
This branch will be merged to master as soon, as Vedder released the FW.

closed 2017-07-02 by RollingGecko