RollingGecko/VescUartControl #14: Support fw 3.28

Teslafly, 2017-09-21:
fw 3.28 is out with the new vesc-tool and it breaks this library. I am going to try and debug it and if i get it working i'll mention it here, but it would be nice to have this working.
right now it is returning messages of length 54, which i'm guessing is some default response considering you're checking for len >55, but haven't gotten as far as to reading the messages.

RollingGecko, 2017-09-22:
Please test this branch
https://github.com/RollingGecko/VescUartControl/tree/VESC6
It has already the changes made on the interface. If it works I will merge it to the master.

Andy

isikxd, 2017-10-14:
By updating to the new library, I get data from UART, but sending my joystick values does not seem to work.

isikxd, 2017-10-14:
Solved!There's a new packet header in the COMM section in VESC6. I just copied the enum in bldc/datatypes.h to your library.

RollingGecko, 2017-10-14:
Can you send me the file or make a pull request?

RollingGecko, 2017-10-14:
Already merged new datatypes.h. But tester required! Please check this: https://github.com/RollingGecko/VescUartControl/issues/15

closed 2017-10-14 by RollingGecko