RollingGecko/VescUartControl #18: VescUartControl for Vesc tool working with Teensy 3.6 Error

jayjaymomo, 2018-02-23:
Hi there, my name is Jessica. 
I was wondering if anyone can provide me with some assistance. I tried running the VescUartControl for the new Firmware V3.28 and I am unfortunately unable to use it on a Teensy 3.6. 

This is the Error that I have received(Seen below)

cannot convert 'usb_serial_class*' to 'HardwareSerial*' for argument '1' to 'void SetDebugSerialPort(HardwareSerial*)'

After hours of research, I have notice that the reason might be because the Teensy runs a different processor than an Arduino. 
Can someone verify this for me?

Thank you, 
Sincerely
Jess :)

fabolhak, 2018-04-30:
Hey, I also had problems getting it to work on our Teensy 3.2. I simply fixed it with changing the type from `HardwareSerial*` to `usb_serial_class*`: 

https://github.com/tum-phoenix/drive_teensy_vesc_control/commit/bce9d1105206404d5bc72050649393fa77210bdf

Teslafly, 2018-10-09:
my branch:
https://github.com/Teslafly/VescUartControl/tree/VESC6_fixedSerial
makes some serial fixups that may help
seems to run on the teensy 3.2 just fine and works with new vesc firmwares (tested on fw3.38)