RollingGecko/VescUartControl #21: Help appreciated for ESP32

flapwazzle, 2018-03-26:
Hi,

I'm trying to get your code working with the ESP32, but I'm having this error:
`\\hrzfs01.hrz.isc.fh-kiel.de\home_st$\selnvett\Arduino\libraries\VescUartControl-master\VescUart.cpp: In function 'int ReceiveUartMessage(uint8_t*)':

\\hrzfs01.hrz.isc.fh-kiel.de\home_st$\selnvett\Arduino\libraries\VescUartControl-master\VescUart.cpp:36:9: error: 'SERIALIO' was not declared in this scope

  while (SERIALIO.available()) {

         ^

\\hrzfs01.hrz.isc.fh-kiel.de\home_st$\selnvett\Arduino\libraries\VescUartControl-master\VescUart.cpp: In function 'int PackSendPayload(uint8_t*, int)':

\\hrzfs01.hrz.isc.fh-kiel.de\home_st$\selnvett\Arduino\libraries\VescUartControl-master\VescUart.cpp:139:21: warning: converting to non-pointer type 'uint8_t {aka unsigned char}' from NULL [-Wconversion-null]

  messageSend[count] = NULL;

                     ^

\\hrzfs01.hrz.isc.fh-kiel.de\home_st$\selnvett\Arduino\libraries\VescUartControl-master\VescUart.cpp:147:2: error: 'SERIALIO' was not declared in this scope

  SERIALIO.write(messageSend, count);

  ^

\\hrzfs01.hrz.isc.fh-kiel.de\home_st$\selnvett\Arduino\libraries\VescUartControl-master\VescUart.cpp: In function 'void SerialPrint(uint8_t*, int)':

\\hrzfs01.hrz.isc.fh-kiel.de\home_st$\selnvett\Arduino\libraries\VescUartControl-master\VescUart.cpp:253:3: error: 'DEBUGSERIAL' was not declared in this scope

   DEBUGSERIAL.print(data[i]);

   ^

\\hrzfs01.hrz.isc.fh-kiel.de\home_st$\selnvett\Arduino\libraries\VescUartControl-master\VescUart.cpp:256:2: error: 'DEBUGSERIAL' was not declared in this scope

  DEBUGSERIAL.println("");

  ^

\\hrzfs01.hrz.isc.fh-kiel.de\home_st$\selnvett\Arduino\libraries\VescUartControl-master\VescUart.cpp: In function 'void SerialPrint(const bldcMeasure&)':

\\hrzfs01.hrz.isc.fh-kiel.de\home_st$\selnvett\Arduino\libraries\VescUartControl-master\VescUart.cpp:261:2: error: 'DEBUGSERIAL' was not declared in this scope

  DEBUGSERIAL.print("avgMotorCurrent: "); DEBUGSERIAL.println(values.avgMotorCurrent);
`

This is my code:

`/*
  Name:    VescUartSample.ino
  Created: 9/26/2015 10:12:38 PM
  Author:  AC
*/

// the setup function runs once when you press reset or power the board
// To use VescUartControl stand alone you need to define a config.h file, that should contain the Serial or you have to comment the line
// #include Config.h out in VescUart.h

//Include libraries copied from VESC
#include "VescUart.h"
#include "datatypes.h"
#include <HardwareSerial.h>

HardwareSerial Serial1(1);

#define DEBUG
unsigned long count;

void setup() {

  //Setup UART port
  Serial1.begin(115200, SERIAL_8N1, 16, 17);
#ifdef DEBUG
  //SEtup debug port
  Serial.begin(115200);
#endif
}

struct bldcMeasure measuredValues;

// the loop function runs over and over again until power down or reset
void loop() {
  //int len=0;
  //len = ReceiveUartMessage(message);
  //if (len > 0)
  //{
  //  len = PackSendPayload(message, len);
  //  len = 0;
  //}

  if (VescUartGetValue(measuredValues)) {
    Serial.print("Loop: "); Serial.println(count++);
    SerialPrint(measuredValues);
  }
  else
  {
    Serial.println("Failed to get data!");
  }

}
`

Do you have ideas? Your help is very much appreciated!

mike1334, 2018-08-28:
flapwazzle, currently I am stuck the same point with the same errors. Have you made any progress since your last post? I've gone through and to the best of my knowledge all files have the includes needed but still get this error i'm baffled. If I do happen to stumble on an answer ill be sure to hit you up!

Teslafly, 2018-10-09:
try this branch
https://github.com/fubarlabs/VescUartControl

otherwise my branch: 
https://github.com/Teslafly/VescUartControl/tree/VESC6_fixedSerial