RollingGecko/VescUartControl #24: Reading rpm out of struct

pumpkinheadnick, 2019-05-12:
Hello, 
I was trying to use your new UART VESC 6 library and I can't figure out how to simply just pull the rpm member out of the struct. I read that something like Serial.println(measuredVal.rpm); should work but all I keep getting returned is 0. 
Any idea what the issue is?

gitcnd, 2019-05-14:
Are you getting anything at all?  Perhaps the call is not working?
I am having a similar problem - I'm getting the data OK, but, VESC 3.56 is not adding the crc and terminating byte (3) on the end - if yours does this too, it's probably causing the VescUart.h to skip returning you the data?

Peemouse, 2019-05-14:
It's maybe due to a buffer overflow. Since the time where Vesc_id has been added to the serialization, the whole packet exceed 63 bytes (default Arduino buffer size).
3 solutions :
1. Switch to COMM_GET_VALUES_SELECTIVE
2. Increase buffer size (a bit a pain)
3. Use @solidgeek 's FW that empty the buffer as the data comes (thus never reach the limit.

In my opinion, solution #1 is the better because it'll stay the same command across the further version, no need to touch your code each time you update.

pumpkinheadnick, 2019-05-14:
I can get all the values to return using the example code but am not able to just pull rpm. 
@gitcnd What is "crc"? 
@Peemouse What is "COMM_GET_VALUES_SELECTIVE"? How can you use that to pull the rpm?

Peemouse, 2019-05-14:
Let me push my fork (which use this command) and you'll be good.
Or easier : use @Solidgeek 's library

gitcnd, 2019-05-14:
@pumpkinheadnick 
> @gitcnd What is "crc"?
Data packets start with a code (2) to say how big the packet length will be (2 means 1 byte length), then a byte to say how long the packet will be, then the packet data itself [which usually begins with a byte to say what data it is - eg - 4 means a reply to COMM_GET_VALUES], then after the data a 2-byte error-checking number called a "CRC" (cyclic redundancy check) and end with a single byte 3.  You probably don't see those things - they get removed for you when you use the packet.

> @Peemouse What is "COMM_GET_VALUES_SELECTIVE"? How can you use that to pull the rpm?
It is the same as COMM_GET_VALUES except you get to list what things you want, and it returns only what you asked for, instead of everything.

RPM is bit number 7 (zero based), so setting the mask to 0x80 will return just RPM to you 

What are you building?  I'm writing a new full-featured remote controller - my code gets RPM fine (I'm not using other peoples libs, which is why I'm messing with crcs etc)

gitcnd, 2019-05-14:
@Peemouse - That definitely sounds like the problem... except I'm using interrupts to fetch serial data the instant it's available on my Arduino Pro-Mini 3.3v (8mhz atmega328p) into a 520-byte buffer, so I am somewhat puzzled that the problem could be on my side - it seems like a bug on the sending side - in the vesc firmware itself.