RollingGecko/VescUartControl #25: Probably does not work with modern firmware

gitcnd, 2019-05-14:
For example - COMM_GET_VALUES in fw 3.55+ returns 65 bytes of data.
This code expects 54 bytes.
The crc etc will of course no longer work...

joseguerr, 2019-05-22:
Could not make it work for FW 3.56. If Anyone knows a way of reading data from the new firmware let me know. I'll post here if i know how to do it. In the meantime what version of FW shoud I use?

Peemouse, 2019-05-22:
Give my fork a try : https://github.com/Peemouse/VescUart
It works for me with FW3.57.

joseguerr, 2019-05-22:
Thanks for quick the replay @Peemouse 

I tried the getPWversion, but no success.. I am using an Arduino mega and used Serial1 as well Rx1 -> 19 and Tx1 ->18. Both baud rates at 115200 and did not change the mask (I know little to nothing of c or c++ but i can learn it). Pins were correctly placed and are giving me 0.5 V for Vesc RX and 3.3V for Vesc Tx. Do you have any idea what can be the problem? 

Thank you once again,
José

Peemouse, 2019-05-22:
Add `  UART.setDebugPort(&Serial);` after `UART.setSerialPort(&Serial1);` in the example file getFWversion.ino. Load it on your Mega and open the Serial monitor.

Peemouse, 2019-05-22:
Well, for the mask, just use the file I linked to you.
I need to add the lines for deserialization so you'll only need to uncomment he ones you're requesting.
Let me push it.

Peemouse, 2019-05-22:
@joseguerr please update your library with the latest version I just pushed.

joseguerr, 2019-05-22:
I have updated the library. When using the getVescValues the rpm and inpVolts are giving me strange values. I do not have the motor connected to the VESC atm.

[image]

Is this a common situation? Thank you once again.

Peemouse, 2019-05-22:
@joseguerr  Could you open an issue on my fork for this ?

flegmatoid, 2019-07-05:
I second Joseguers conclusion. v3.58 returns huge values in both idle and running cases. Something about the packet format got changed.

joseguerr, 2019-07-06:
The code provided is working properly. At the time I was having issues because the grounds were not properly connected. Also, I was using Arduino Mega and one thing it worked for me was to use TX and RX 3, since 1 and 2 were not giving me any decent output. Go check Peemuse's issues, I believe that is the place where the full discussion is. Cheers